WLAN-Pi/wlanpi-profiler #9: Use database to store results rather than flat files

joshschmelzle, 2020-05-21:
- implement code to work with a database to store analysis results, this is either in addition to, or rather than output to flat files.
- need to consider ease of use.

wifinigel, 2020-05-21:
Suggestions:

1. As discussed, implement DB via SqlAlchemy to allow easy re-structure of DB as new capabilities added (e.g. 11ax capabilities).
2. DB Fields in addition to derived capabilities:
  a. profile ID (unique index)
  b. profile timestamp
  c. client MAC
  d. MAC OUI manuf lookup

joshschmelzle, 2021-11-24:
As an update, one of the things that was implemented (not sure which commit) is JSON output along with the `.txt` and `.pcap` files. The plan is to ship something like this to a public/centralized web app if time permits for development.

It looks like this:

```
{"mac": "92-80-6b-37-52-61", "is_laa": true, "manuf": "Apple", "band": 5, "capture_channel": 36, "listen_only": false, "features": {"dot11k": 1, "dot11r": 1, "dot11v": 1, "dot11w": 1, "dot11n": 1, "dot11n_nss": 2, "dot11ac": 0, "dot11ac_nss": 2, "dot11ac_160_mhz": 0, "dot11ac_mcs": "0-9", "dot11ac_su_bf": 1, "dot11ac_mu_bf": 0, "dot11ax": 1, "dot11ax_nss": 2, "dot11ax_mcs": "0-11", "dot11ax_twt": 0, "dot11ax_uora": 0, "dot11ax_bsr": 1, "dot11ax_punctured_preamble": 0, "dot11ax_he_su_beamformer": 0, "dot11ax_he_su_beamformee": 1, "dot11ax_he_er_su_ppdu": 0, "dot11ax_six_ghz": 0, "dot11ax_160_mhz": 0, "six_ghz_operating_class_supported": 0, "max_power": 21, "min_power": -7, "supported_channels": [36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165]}, "pcap": "\"AAAeAC5AAKAgCACgIAgAAAAMPBRAAbsAAAC7ALUBAAA8ALA5Vo1peJKAazdSYbA5Vo1peABrEREUAAAHV0xBTiBQaQEIjBKYJLBIYGwhAvkVJAokBDQEZAyVBKUBMBQBAAAPrAQBAAAPrAQBAAAPrASMAEYFMQgBAAA2A0XCAC0abwAb//8AAAAAAAAAAAAAAAAAAAAAAAAAAAB/CAAACAAAAABAvwwycIEP+v8AAPr/AADHARD/HCMBCAgAAIBEMAIAHQCfCAAMAPr/+v85HMdxHAfdCwAX8goAAQQAAAAA3QUAkEwEB90KABAYAgAAEAAAAN0HAFDyAgABAA==\"", "schema_version": 1, "profiler_version": "1.0.9-1"}
```