WLAN-Pi/wlanpi-profiler #11: daemonize profiler so that start/stop/restart works.

joshschmelzle, 2020-05-27:
Perhaps start here: https://stackoverflow.com/questions/16420092/how-to-make-python-script-run-as-service

danielmundi, 2020-05-27:
@joshschmelzle, you can actually just create a systemd unit service for your script without daemonizing it.
You set ExecStart attribute for the command to execute profiler and systemd will take care of stop and restart.

I can help you with this service if you need.

joshschmelzle, 2020-05-27:
@danielmundi thanks for the pointer! Let me take a stab at it and I'll pass it by you.

joshschmelzle, 2020-05-28:
ended up with the following for `profiler.service`:

```
[Unit]
Description=profiler2 service
After=multi-user.target

[Service]
Type=idle
ExecStart=/usr/local/bin/profiler2
```

thanks @danielmundi for the help.

closed 2020-05-28 by joshschmelzle