WLAN-Pi/wlanpi-profiler #42: updating scapy dependency from 2.4.3 to 2.4.4 breaks profiler

joshschmelzle, 2020-09-03:
```
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/wlanpi/profiler2/profiler2/fakeap.py", line 119, in __init__
    self.every(self.beacon_interval, self.beacon)
  File "/opt/wlanpi/profiler2/profiler2/fakeap.py", line 125, in every
    task()
  File "/opt/wlanpi/profiler2/profiler2/fakeap.py", line 154, in beacon
    self.l2socket.send(frame)
  File "/usr/local/lib/python3.7/dist-packages/scapy/arch/linux.py", line 476, in send
    return SuperSocket.send(self, x)
  File "/usr/local/lib/python3.7/dist-packages/scapy/supersocket.py", line 71, in send
    sx = raw(x)
  File "/usr/local/lib/python3.7/dist-packages/scapy/compat.py", line 53, in raw
    return bytes(x)
  File "/usr/local/lib/python3.7/dist-packages/scapy/packet.py", line 498, in __bytes__
    return self.build()
  File "/usr/local/lib/python3.7/dist-packages/scapy/packet.py", line 618, in build
    p = self.do_build()
  File "/usr/local/lib/python3.7/dist-packages/scapy/packet.py", line 600, in do_build
    pkt = self.self_build()
  File "/usr/local/lib/python3.7/dist-packages/scapy/packet.py", line 581, in self_build
    p = f.addfield(self, p, val)
  File "/usr/local/lib/python3.7/dist-packages/scapy/fields.py", line 253, in addfield
    return self.fld.addfield(pkt, s, val)
  File "/usr/local/lib/python3.7/dist-packages/scapy/fields.py", line 151, in addfield
    return s + self.struct.pack(self.i2m(pkt, val))
struct.error: ubyte format requires 0 <= number <= 255
```

joshschmelzle, 2020-09-03:
scapy did not like the way we were building the radiotap header. this has been fixed in dev. https://github.com/joshschmelzle/profiler2/commit/ad8af90c0328d9261043ce33f3f46518ccade976

closed 2020-09-03 by joshschmelzle

AmandaYaoYao, 2020-10-21:
> ```
> Traceback (most recent call last):
>   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
>     self.run()
>   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, [...]

I meet the same problem and just can rollback the version to the 2.4.3

joshschmelzle, 2020-10-21:
@AmandaYaoYao this should be resolved in the latest version of profiler.