WLAN-Pi/wlanpi-profiler #106: Is this working on Kali Linux?

drschnalli, 2022-03-04:
Hey there... i just installed profiler via pipx. It worked fine.. Its also starting and showing me that its using wlan0mon (i got a wifi ax200...) but i cannot find this network... is it even possible to run it on amd64 kali linux? What might i do wrong?

joshschmelzle, 2022-03-04:
@drschnalli i think it _should_ work, but i have not tested on Kali. i've previously tested on Debian/Ubuntu x86_64 and it worked. you may have to run it with the no bpf filters option.

can you try running `sudo tcpdump -i wlan0mon` while profiler is running to check for beacons/probes/traffic?

another thing to try is running profiler with the `--debug` option like `sudo profiler --debug` which should print more infos to the terminal.

drschnalli, 2022-03-04:
I will have a look with tcpdump.

by the way is it a problem that it shows: no ir found in iw channel?

joshschmelzle, 2022-03-04:
If you're using an AX200 or AX210, profiler will check for No IR and then do a scan to attempt to find a CC in a nearby .11d IE from beacon to enable.

AX2xx uses iwlwifi which has an embedded reg dom (doesn't get regdom from iw or cfg80211). So, driver may only be looking for US CCs. If you're outside of the US or don't have a US AP nearby, it might not work.

If the channel says No IR after profiler starts, profiler will not be able to TX on that channel. So, you may try setting the channel to a 2.4 GHz channel if that is the case: `sudo profiler -c 1`.

joshschmelzle, 2022-03-04:
Another option if Wi-Fi scan with AX200 does not enable TX in your location. You can also try any mt76x2u USB NIC which will use CC set with `iw reg set XX` for 5 GHz.

drschnalli, 2022-03-04:
Okay when i run the tcpdump command on that interface the window shows many many many traffic... so something seems to happen. I can confirm its working with another adapter  ALFA Network AWUS036NHA. A very old usb dongle :D... for the AX200 i will have a look for your other suggestions!

joshschmelzle, 2022-03-04:
We don't need anything special for the profiler "AP" because we're just forging a fake beacon anyways. So, the AWUS036NHA is sufficient because profiler will append .11ac / .11he IEs regardless.

drschnalli, 2022-03-04:
Okay it works when i set the Channel to 1!!! Thats so cool.. thank you! Is there a way to Simulate being in US ? or Changing something?

drschnalli, 2022-03-04:
Or do i not need to be in 5ghz to get information about 802.11r of the connected client? or other informations like ax, ac and so on ? is 2,4 fully sufficient?

joshschmelzle, 2022-03-04:
Client won't reveal 5 GHz infos when connecting to 2.4 GHz. 

We don't know of any software tricks for setting regdom for iwlwifi/Intel AX2xx other than placing a US AP nearby. @jiribrejcha literally used a huge cooking pot in one of his tests.

[image]

jiribrejcha, 2022-03-04:
😄

drschnalli, 2022-03-05:
Thats an awesome idea :D what were the results? :D

closed 2022-09-09 by joshschmelzle