WLAN-Pi/wlanpi-profiler #200: Usage documentation is outdated after 1.0.17 and 1.0.18 released

joshschmelzle, 2024-09-19:
From the docs:

```
$ profiler -h
usage: profiler [-h] [-c CHANNEL | -f FREQUENCY] [-i INTERFACE] [-s SSID] [--config FILE]
                [--files_path PATH] [--hostname_ssid] [--debug] [--logging [{debug,warning}]]
                [--noprep] [--noAP] [--no11r] [--no11ax] [--oui_update] [--read PCAP] 
                [--no_bpf_filters] [--list_interfaces] [--version]

options:
  -h, --help            show this help message and exit
  -c CHANNEL            set the channel to broadcast on
  -f FREQUENCY          set the frequency to broadcast on
  -i INTERFACE          set network interface for profiler
  -s SSID               set profiler SSID name
  --config FILE         customize path for configuration file 
                            (default: /etc/wlanpi-profiler/config.ini)
  --files_path PATH     customize default directory where analysis is saved on local system
                            (default: /var/www/html/profiler)
  --hostname_ssid       use the WLAN Pi's hostname as SSID name (default: False)
  --debug               enable debug logging output
  --logging [{debug,warning}]
                        change logging output
  --noprep              disable interface preperation (default: False)
  --noAP                enable Rx only mode (default: False)
  --no11r               turn off 802.11r Fast Transition (FT) reporting
  --no11ax              turn off 802.11ax High Efficiency (HE) reporting
  --oui_update          initiates update of OUI database (requires Internet connection)
  --read PCAP           read and analyze association request frames from pcap
  --no_bpf_filters      removes BPF filters from sniffer() but may impact profiler performance
  --list_interfaces     print out a list of interfaces with an 80211 stack
  --version, -V         show program's version number and exit
```

Needs updated accordingly:

```
usage: profiler [-h] [-c CHANNEL | -f FREQUENCY] [-i INTERFACE] [-s SSID] [--config FILE] [--files_path PATH] [--hostname_ssid] [--debug] [--noprep]
                   [--noAP] [--no11r] [--no11ax] [--no11be] [--noprofilertlv] [--wpa3_personal_transition | --wpa3_personal] [--oui_update] [--read PCAP]
                   [--no_bpf_filters] [--list_interfaces] [--version]

wlanpi-profiler is an 802.11 client capabilities profiler. If installed via apt package manager, read the manual with: man wlanpi-profiler

optional arguments:
  -h, --help            show this help message and exit
  -c CHANNEL            set the channel to broadcast on
  -f FREQUENCY          set the frequency to broadcast on
  -i INTERFACE          set network interface for profiler
  -s SSID               set profiler SSID name
  --config FILE         customize path for configuration file (default: /etc/wlanpi-profiler/config.ini)
  --files_path PATH     customize default directory where analysis is saved on local system (default: /var/www/html/profiler)
  --hostname_ssid       use the WLAN Pi's hostname as SSID name (default: False)
  --debug               enable debug logging output
  --noprep              disable interface preperation (default: False)
  --noAP                enable Rx only mode (default: False)
  --no11r               turn off 802.11r Fast Transition (FT) reporting
  --no11ax              turn off 802.11ax High Efficiency (HE) reporting
  --no11be              turn off 802.11be Extremely High Throughput (EHT) reporting
  --noprofilertlv       disable generation of Profiler specific vendor IE
  --wpa3_personal_transition
                        enable WPA3 Personal Transition in the RSNE for 2.4 / 5 GHz
  --wpa3_personal       enable WPA3 Personal only in the RSNE for 2.4 / 5 GHz
  --oui_update          initiates update of OUI database (requires Internet connection)
  --read PCAP           read and analyze association request frames from pcap
  --no_bpf_filters      removes BPF filters from sniffer() but may impact profiler performance
  --list_interfaces     print out a list of interfaces with an 80211 stack
  --version, -V         show program's version number and exit
```

lisamessier, 2024-09-19:
https://github.com/WLAN-Pi/wlanpi-profiler/pull/201

joshschmelzle, 2024-09-25:
Closed by #201. Thanks @lisamessier

closed 2024-09-25 by joshschmelzle